IGSIO/SlicerIGSIO #15: SlicerIGSIO extenssion not show up in Slicer version 4.13.0 revision 30557 built 2022-01-26

1939938853, 2022-01-27:
Hello,

SlicerIGSIO extension not show up in Slicer version 4.13.0 revision 30557 built 2022-01-26. I manually installed the 30117-win-amd64-SlicerIGSIO-git8f61201-2021-07-19.zip and run Slicer got the following errors.

Thanks

DLL load failed while importing vtkSlicerMetafileImporterModuleLogicPython: The specified module could not be found.
DLL load failed while importing vtkSlicerSequenceIOModuleLogicPython: The specified module could not be found.
DLL load failed while importing vtkSlicerVideoUtilModuleLogicPython: The specified module could not be found.
DLL load failed while importing vtkSlicerVolumeReconstructionModuleLogicPython: The specified module could not be found.
DLL load failed while importing vtkSlicerSequenceIOModuleMRMLPython: The specified module could not be found.
DLL load failed while importing vtkSlicerVolumeReconstructionModuleMRMLPython: The specified module could not be found.

jamesobutler, 2022-01-27:
Installing a ZIP file that doesn't come from a corresponding version of Slicer isn't expected to work. SlicerIGSIO is currently not building successfully in latest Slicer preview which is why it is not available.

You can try an older version of Slicer which you can download from https://slicer.cdash.org/viewFiles.php?buildid=2559576.

SlicerIGSIO not building successfully because of the following only supporting up to Visual Studio 2019 (v142 toolset). Current Slicer preview builds are using Visual Studio 2022 (v143 toolset).
https://github.com/IGSIO/SlicerIGSIO/blob/d043a22df959959c9cbb32157976506a1b0c1122/SuperBuild/External_VP9.cmake#L83-L89

xref https://github.com/Slicer/Slicer/issues/6140

Sunderlandkyl, 2022-01-28:
SlicerIGSIO is available again in the latest nightly.

closed 2022-01-28 by Sunderlandkyl