IGSIO/SlicerIGSIO #18: CMake variable EXTENSION_WC_REVISION is empty !

whucsu, 2022-04-11:
Slicer 4.11 compiled successfully.
And the IGSIO compiled successfully.
But as to the SlicerIGSIO compiling,  only one error :
CMake Error at C:/Slicer/SlicerSource/Extensions/CMake/SlicerFunctionGenerateExtensionDescription.cmake:70 (message):
8>  CMake variable EXTENSION_WC_REVISION is empty !
thanks for any help or suggestion.

Sunderlandkyl, 2022-04-11:
Is the SlicerIGSIO code not under version control?

If it isn't, then you can define the variables manually in CMake. See post here:
https://discourse.slicer.org/t/3d-slicer-release-build-error-variable-slicer-wc-last-changed-date-is-expected-to-be-defined-in-slicerpackageanduploadtarget/20984/23?u=sunderlandkyl

whucsu, 2022-04-12:
thanks, I  searched that issue. I will do as that suggetstion.

whucsu, 2022-04-12:
> Is the SlicerIGSIO code not under version control?
>
> If it isn't, then you can define the variables manually in CMake. See post here: [...]

dose it mean that I need to re-compile the Slicer and add that TWO cmake variables ?