IGSIO/SlicerIGSIO #21: Specify non-changing IGSIO version

jamesobutler, 2022-04-19:
When attempting to integrate SlicerIGSIO as part of a Slicer custom application, I was using SlicerIGSIO at specific commit d38a5a239645b6ad38e49c573dee51e8ae7c2344. This commit specifies to use IGSIO's `master` branch https://github.com/IGSIO/SlicerIGSIO/blob/d38a5a239645b6ad38e49c573dee51e8ae7c2344/SuperBuild/External_IGSIO.cmake#L27. This has led to issues as since early February IGSIO has an ITK dependency and needed corresponding change in 6815fb30ba4cb32a710d9a68e58e24527696a51b which came after the commit I used.

Can we set SlicerIGSIO to use a non-changing git tag of IGSIO so that a specific version of this extension can be included in a Slicer custom application?

cc: @Sunderlandkyl

Sunderlandkyl, 2022-04-19:
Yeah, you can specify "${CMAKE_PROJECT_NAME}_IGSIO_GIT_TAG" to lock IGSIO to a specific revision.

jamesobutler, 2022-04-20:
Thanks for the help! I appear to have been successful with the following code below in the main CMakeLists.txt of my Slicer custom application. I confirmed that at "C:\MyCustomApp-bin\IGSIO" that it was using the newly specified git tag. If you agree that this is the correct solution please close this issue.

```diff
#-----------------------------------------------------------------------------
# SlicerIGSIO
set(extension_name "SlicerIGSIO")
set(${extension_name}_SOURCE_DIR "${CMAKE_BINARY_DIR}/${extension_name}")
+ExternalProject_SetIfNotDefined(
+  ${CMAKE_PROJECT_NAME}_IGSIO_GIT_TAG
+  "470b1553768981f3a86b9a79bf0f9016bebe74ce"
+  QUIET
+  )
FetchContent_Populate(${extension_name}
 SOURCE_DIR     ${${extension_name}_SOURCE_DIR}
 GIT_REPOSITORY ${EP_GIT_PROTOCOL}://github.com/IGSIO/${extension_name}.git
 GIT_TAG        762d256a24e10e740d307022cac098cd3032c09a
 GIT_PROGRESS   1
 QUIET
 )
list(APPEND Slicer_EXTENSION_SOURCE_DIRS ${${extension_name}_SOURCE_DIR})
ExternalProject_Add_Dependencies(IGSIO
  DEPENDS
    VTK
    VP9
    YASM
  )
# Slicer becomes needs vtkAddon
ExternalProject_Add_Dependencies(YASM
  DEPENDS
    python
  )
```

closed 2022-04-21 by jamesobutler