IGSIO/SlicerIGSIO #24: Build error in latest Slicer-5.1

lassoan, 2022-10-17:
Slicer's VTK has been updated and it no longer contains vtkSimpleCriticalSection. It seems it has been replaced by std::mutex.

SlicerIGSIO depends on vtkSimpleCriticalSection to implement vtkRecursiveCriticalSection and so its build fails and all extensions depending on it fails, too.

The problem may be resolved by either replacing vtkSimpleCriticalSection by std::mutex or [std::recursive_mutex](https://en.cppreference.com/w/cpp/thread/recursive_mutex).

lassoan, 2022-10-17:
@Sunderlandkyl could you look into this?

Sunderlandkyl, 2022-10-17:
Sure, I'll look at both issues.

Sunderlandkyl, 2022-10-19:
Fixed in https://github.com/IGSIO/IGSIO/commit/44f68876d956ea51b076b4ccea3f7850a42a4d03 and https://github.com/IGSIO/IGSIO/commit/131531cd16e780a18c278aa6b572cf863513d6db

closed 2022-10-19 by Sunderlandkyl