IGSIO/SlicerIGSIO #27: Improve inclusion of CL headers.

chir-set, 2023-05-01:
Today's build of SlicerIGSIO failed on Arch Linux, while it used to build seamlessly and there does not seem to have been commits since 10/12/2022.

CL/cl.hpp could not be found.

A quick and dirty fix has been editing IGSIO/CMakeLists.txt in the build tree as such :

`include_directories(/opt/rocm/include /opt/rocm/opencl/include)`

A better fix would be to allowing passing an OpenCL_DIR when configuring. As stated above, Arch Linux does not install these files in /usr/include.

Just a ping for a future improvement if you find it important.

Regards.

lassoan, 2023-05-02:
I don't see any errors on the dashboards, neither Preview nor Stable, on any platforms. We recently changed defaults to enable OpenCL acceleration. If you don't have OpenCL on your computer then you may disable it (it is only used for faster volume reconstruction).

@Sunderlandkyl could you check if we could pass on OpenCL folder(s)?

Sunderlandkyl, 2023-05-02:
Yeah, we can add the option to SlicerIGSIO.

SlicerIGSIO_USE_GPU should only be enabled by default if find_package(OpenCL) is successful.
Does find_package(OpenCL) not find the correct paths?

chir-set, 2023-05-02:
> Does find_package(OpenCL) not find the correct paths?

That seems to be the case.

If I specify 

> -DOpenCL_DIR:PATH=/opt/rocm/opencl

 and configure, the result is : 

```
CMake Warning:
  Manually-specified variables were not used by the project:

    OpenCL_DIR
```

At build : 

```
In file included from /home/arc/src/SlicerIGSIO-SuperBuild9/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolume.cxx:59:
/home/arc/src/SlicerIGSIO-SuperBuild9/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:61:10: fatal error: 'CL/cl.hpp' file not found
#include <CL/cl.hpp>

```

chir-set, 2023-05-14:
Actually, I found an Arch opench-headers package that installs the headers in /usr/include. So this issue becomes moot, closing and sorry for the noise.

closed 2023-05-14 by chir-set