IGSIO/SlicerIGSIO #28: [Windows][VS22] Configuring fails

adamrankin, 2023-10-03:
After building latest Slicer, configuring of inner project fails

```
CMake Error at C:/d/Slcr/S4/Extensions/CMake/SlicerBlockAdditionalLauncherSettings.cmake:97 (string):
  string sub-command REPLACE requires at least four arguments.
```

Offending code is 
``` cmake
  # External projects - library paths
  foreach(varname IN LISTS ${SUPERBUILD_TOPLEVEL_PROJECT}_EP_LABEL_LIBRARY_PATHS_LAUNCHER_BUILD)
    set(value ${${varname}})
    string(REPLACE "<CMAKE_CFG_INTDIR>" "\${CMAKE_CFG_INTDIR}" value ${value})
    list(APPEND EXTENSION_LIBRARY_PATHS_BUILD ${value})
  endforeach()
```

Do any of you guys know of a change in the Slicer level variable that might break this?

Sunderlandkyl, 2023-10-03:
This looks to be this change in Slicer: https://github.com/Slicer/Slicer/commit/b1efab8a65c097621aad2a2eaf2ba1b17ebad212

There is only one variable in "inner_EP_LABEL_LIBRARY_PATHS_LAUNCHER_BUILD": "VP9_LIBRARY_PATHS_LAUNCHER_BUILD". The value  of VP9_LIBRARY_PATHS_LAUNCHER_BUILD is empty, causing the error in string(REPLACE ...).

Sunderlandkyl, 2023-10-03:
I'm testing a fix now.

commit 1a89776c (2023-10-03): COMP: Remove empty LAUNCHER CMake variables in External_VP9.cmake

Due to recent changes in Slicer (https://github.com/Slicer/Slicer/commit/b1efab8a65c097621aad2a2eaf2ba1b17ebad212), configuration will fail if a launcher variable (ex. ${proj}_LIBRARY_PATHS_LAUNCHER_BUILD) was defined, and is empty. Fixed the issue by not defining the LIBRARY_PATHS_LAUNCHER_BUILD variables that will be empty.

Fixes #28
closed 2023-10-03 by Sunderlandkyl