IGSIO/SlicerIGSIO #29: Error building on Ubuntu 22.04

bluewish07, 2024-09-16:
System: Ubuntu 22.04
I have 3DSlicer built from source successfully and the build products reside in <working directory>/Slicer-SuperBuild-Release/

Here's the instructions that I followed to build
```
$ cd <working directory>/modules
$ git clone https://github.com/IGSIO/SlicerIGSIO
$ mkdir SlicerIGSIO-build
$ cd SlicerIGSIO-build
$ cmake -DCMAKE_BUILD_TYPE:STRING=Release -DSlicer_DIR:PATH=<working directory>/Slicer-SuperBuild-Release/Slicer-build ../SlicerIGSIO
$ make -j4
```

Here's the error that I got:

```
[ 86%] Built target vtkIGSIOCalibration
In file included from /home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:61,
                 from /home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolume.cxx:63:
/usr/local/cuda/include/CL/cl.hpp:437:112: note: ‘#pragma message: opencl.hpp: CL_HPP_TARGET_OPENCL_VERSION is not defined. It will default to 300 (OpenCL 3.0)’
  437 | ET_OPENCL_VERSION is not defined. It will default to 300 (OpenCL 3.0)")
      |                                                                       ^

In file included from /home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolume.cxx:63:
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h: In function ‘void vtkOpenCLInsertSlice(vtkIGSIOPasteSliceIntoVolumeInsertSliceParams*, int, vtkOpenCLContext**)’:
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:534:7: error: ‘size_t’ is not a member of ‘cl’; did you mean ‘Size’?
  534 |   cl::size_t<3> buffer_origin;
      |       ^~~~~~
      |       Size
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:534:17: error: ‘buffer_origin’ was not declared in this scope; did you mean ‘cl_buffer_region’?
  534 |   cl::size_t<3> buffer_origin;
      |                 ^~~~~~~~~~~~~
      |                 cl_buffer_region
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:535:7: error: ‘size_t’ is not a member of ‘cl’; did you mean ‘Size’?
  535 |   cl::size_t<3> in_origin;
      |       ^~~~~~
      |       Size
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:535:17: error: ‘in_origin’ was not declared in this scope; did you mean ‘inOrigin’?
  535 |   cl::size_t<3> in_origin;
      |                 ^~~~~~~~~
      |                 inOrigin
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:536:7: error: ‘size_t’ is not a member of ‘cl’; did you mean ‘Size’?
  536 |   cl::size_t<3> in_region;
      |       ^~~~~~
      |       Size
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:536:17: error: ‘in_region’ was not declared in this scope; did you mean ‘inOrigin’?
  536 |   cl::size_t<3> in_region;
      |                 ^~~~~~~~~
      |                 inOrigin
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:537:7: error: ‘size_t’ is not a member of ‘cl’; did you mean ‘Size’?
  537 |   cl::size_t<3> out_origin;
      |       ^~~~~~
      |       Size
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:537:17: error: ‘out_origin’ was not declared in this scope
  537 |   cl::size_t<3> out_origin;
      |                 ^~~~~~~~~~
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:538:7: error: ‘size_t’ is not a member of ‘cl’; did you mean ‘Size’?
  538 |   cl::size_t<3> out_region;
      |       ^~~~~~
      |       Size
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:538:17: error: ‘out_region’ was not declared in this scope
  538 |   cl::size_t<3> out_region;
      |                 ^~~~~~~~~~
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:539:7: error: ‘size_t’ is not a member of ‘cl’; did you mean ‘Size’?
  539 |   cl::size_t<3> acc_origin;
      |       ^~~~~~
      |       Size
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:539:17: error: ‘acc_origin’ was not declared in this scope
  539 |   cl::size_t<3> acc_origin;
      |                 ^~~~~~~~~~
[... 34 lines cut ...]
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:693:17: error: ‘acc_origin’ was not declared in this scope
  693 |   cl::size_t<3> acc_origin;
      |                 ^~~~~~~~~~
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:694:7: error: ‘size_t’ is not a member of ‘cl’; did you mean ‘Size’?
  694 |   cl::size_t<3> acc_region;
      |       ^~~~~~
      |       Size
/home/diana/Source/modules/SlicerIGSIO-build/IGSIO/VolumeReconstruction/vtkIGSIOPasteSliceIntoVolumeHelperOpenCL.h:694:17: error: ‘acc_region’ was not declared in this scope
  694 |   cl::size_t<3> acc_region;
      |                 ^~~~~~~~~~
make[8]: *** [VolumeReconstruction/CMakeFiles/vtkVolumeReconstruction.dir/build.make:76: VolumeReconstruction/CMakeFiles/vtkVolumeReconstruction.dir/vtkIGSIOPasteSliceIntoVolume.cxx.o] Error 1
make[7]: *** [CMakeFiles/Makefile2:251: VolumeReconstruction/CMakeFiles/vtkVolumeReconstruction.dir/all] Error 2
make[6]: *** [Makefile:136: all] Error 2
make[5]: *** [CMakeFiles/inner-build.dir/build.make:86: IGSIO-prefix/src/inner-build-stamp/inner-build-build] Error 2
make[4]: *** [CMakeFiles/Makefile2:111: CMakeFiles/inner-build.dir/all] Error 2
make[3]: *** [Makefile:91: all] Error 2
make[2]: *** [CMakeFiles/IGSIO.dir/build.make:86: IGSIO-prefix/src/IGSIO-stamp/IGSIO-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:897: CMakeFiles/IGSIO.dir/all] Error 2
make: *** [Makefile:101: all] Error 2

```

I saw a previously closed issue which might be related: https://github.com/IGSIO/SlicerIGSIO/issues/27. I tried to see if I could get OpenCL in shape on my system by manually apt installing it. I've also tried leaving `SlicerIGSIO_USE_GPU_DEFAULT` to be off. With those efforts I still get the same errors. 

Would appreciate any advice!

lassoan, 2024-09-17:
Does changing `cl::size_t` to `size_t` fix the problem?

bluewish07, 2024-09-18:
I don't think I can do that since I can't do size_t<3>, i.e. c++ native size_t is not a template. And I can't drop <3> because there's indexing code into these variables in the following lines.

adamrankin, 2024-09-18:
Could try `std::array<size_t, 3>`?

bluewish07, 2024-09-18:
Facepalm you are right. And yes that worked. Any chance to change that in a way that would work for everyone (as opposed to just me modifying it locally)? I also don't seem to find that header file in this repo. Is it pulled from another repo?