IGSIO/SlicerIGSIO #30: [Windows][VS22] vtkAddon.lib link error while building

BowMonk, 2025-02-27:
Hello everyone :]

I am trying to build SlicerIGSIO (to then build SlicerIGT), however I ran into a build issue.
I am getting this: 

```
LINK : fatal error LNK1181: cannot open input file 'vtkAddon.lib' [D:\SIGSS\IGSIO-build\inner-build\Codecs\vtkIGSIOCodecs.vcxproj]
```
Where SIGSS is my source folder.

I am not that experienced with CMake, however when trying to build SlicerOpenIGTLink (which I also need, and have separate build issues with), it was able to find vtkAddon with no issue, so I don't think there's something wrong with my file positioning. 

Could it be that the CMake file is missing some paths? I am a bit clueless on this front, so any help would be great.

Thank you in advance.

BowMonk, 2025-02-27:
The issue was that I was not building ALL_BUILD, just a smaller solution file.

closed 2025-02-27 by BowMonk

lassoan, 2025-02-28:
Thanks for the update!